fpistm/Arduino_Core_STM32F1 #1: Need support for other Variants

RickKimball, 2016-09-18:
It would be nice to have support for other variants. I think the BluePill would be a good one to try at first.

rogerclarkmelbourne, 2016-10-29:
Still in progress ;-)

rogerclarkmelbourne, 2016-11-14:
Some progress has been made, we now have a Blue Pill and Nucleo F103RB
and to some extents a Value Line STM Discovery F100 (but the static lib won't compile for this at the moment due to inclusion of USB files in the makefile - I suspect a separate makefile (.mk) is required for this variant

rogerclarkmelbourne, 2016-11-18:
STM32 VL Discovery should now be working
Maple mini has now been added (though only the new bootloader version works - not sure why this is, and dont have time to debug this at the moment)

closed 2017-10-24 by fpistm