fpistm/Arduino_Core_STM32F1 #10: STM32F103RB-Nucleo runs at 64MHz instead of 72MHz

ProjectsByJRP, 2016-11-12:
HAL_RCC_GetSysClockFreq() returns 64000000

rogerclarkmelbourne, 2016-11-12:
Thanks.

I suspect the wrong PLL value may have been used. There was a similar issue on the Nucleo L4

danieleff, 2016-11-12:
I think Nucleo boards do not come with an installed external oscillator. See [UM1742](http://www.st.com/content/ccc/resource/technical/document/user_manual/98/2e/fa/4b/e0/82/43/b7/DM00105823.pdf/files/DM00105823.pdf/jcr:content/translations/en.DM00105823.pdf), bottom of first image X3 is missing, and section  6.7.1 "HSE oscillator on-board from X3 crystal (not provided)"
Without HSE, the STM32F103RB is limited to 64MHz

rogerclarkmelbourne, 2016-11-12:
Thanks Daniel

Now you remind me, I remember that it does not have the external crystal.

I just looked on MBED and it does list the speed as 72Mhz

https://developer.mbed.org/platforms/ST-Nucleo-F103RB/#nucleo-features

danieleff, 2016-11-12:
In the [reference manual](http://www.st.com/content/ccc/resource/technical/document/reference_manual/59/b9/ba/7f/11/af/43/d5/CD00171190.pdf/files/CD00171190.pdf/jcr:content/translations/en.CD00171190.pdf#92), page 92, under clock tree: "When the HSI is used as a PLL clock input, the maximum system clock frequency that can be achieved is 64 MHz."
And that is the max in STM32CubeMX too.

rogerclarkmelbourne, 2016-11-12:
Thanks Daniel

@ProjectsByJRP

See above... 64Mhz is the correct speed unless you have added an external crystal

If your board has an external crystal, its not technically supported by STM's own Nucleo code
However as the community is adding new variants, it would be possible to potentially add a menu that allowed for the selection of the default HSI osc with 64Mhz operation and 8Mhz HSE with 72Mhz operation

ProjectsByJRP, 2016-11-12:
my apologies. that seems obvious now that i look at code checking HSE instead of HSI. i am sorry if i caused a headache.

rogerclarkmelbourne, 2016-11-13:
No worries.
Its actually useful that you highlighted this because we should add support for Nucleo F103's with external Crystal at some time in the future

RickKimball, 2016-11-14:
https://github.com/RickKimball/Arduino_Core_STM32F1/blob/master/variants/STM32F103C8-BluePill/hw_config.c#L114 .. i had fixed that the first time around

closed 2017-10-24 by fpistm