fpistm/Arduino_Core_STM32F1 #14: Convoluted code to get port and pin from pin number

rogerclarkmelbourne, 2016-11-13:
Code to get port and in seems time consuming and overly complex.

``  //find the pin.
  for(i = 0; i < NB_PIN_DESCRIPTIONS; i++) {
    if(g_APinDescription[i].arduino_id == ulPin) {
      g_digPinConfigured[ulPin] = g_APinDescription[i];
      g_digPinConfigured[ulPin].configured = true;
      break;
    }
  }
``

Simple array lookup would probably surfice

danieleff, 2016-11-16:
For this the value in arduino_id in g_APinDescription has to be the same as its array index.
In BluePill and VLDiscovery variants, this is true.
In Nucleo, it is not even unique (lots of ARDUINO_PIN_EXT). Here is a commit change that: 3095d520d02

After that, if the rows are rearranged (`{ PA3,   GPIO_PIN_3,  GPIOA ...}, { PA2,   GPIO_PIN_2,  GPIOA, ...}, { PA10,  GPIO_PIN_10, ...} ...`) the port and pin search can be an array lookup. And g_digPinConfigured can be removed / substituted by g_APinDescription altogether.

fpistm, 2017-10-24:
Pin mapping enhanced in https://github.com/stm32duino/Arduino_Core_STM32

closed 2017-10-24 by fpistm