fpistm/Arduino_Core_STM32F1 #21: Use pin enum names in g_APinDescription in BluePill variant.cpp

danieleff, 2016-11-15:
The BluePill board does not have pin numbers written on the board, use pin names in its variant.cpp:
`extern const PinDescription g_APinDescription[]=`
`{`
`{ PA0, GPIO_PIN_0, GPIOA, ...`
`{ PA1, GPIO_PIN_1, GPIOA, ...`
`{ PA2, GPIO_PIN_2, GPIOA, ...`
`...`

rogerclarkmelbourne, 2016-11-15:
Is this just for clarity in the code.

I already use pin names in my BluePill test code  e.g.

`  pinMode(PB12,OUTPUT);`

and it works fine.

I have posted a separate issue about the complex way that pin name to Port / Pin is done (using a for loop), which seems a waste of time and code when I think it can be accomplished just using an array lookup

rogerclarkmelbourne, 2016-11-15:
Edit.

I see what you mean

It still uses ARDUINO_PIN_A0 etc. I presume this had been changed to 0

I'll fix this

commit 54913088 (2016-11-15): Fixed #21
closed 2016-11-15 by rogerclarkmelbourne