fpistm/Arduino_Core_STM32F1 #23: Global structs consume a lot of RAM

rogerclarkmelbourne, 2016-11-15:
See http://stm32duino.com/viewtopic.php?f=16&t=1553&p=20098#p20058

danieleff, 2016-11-16:
`g_analog_config` is 2KB in RAM. It is only compiled in because [pinMode() calls pwm_stop() and dac_stop()](https://github.com/stm32duino/Arduino_Core_STM32F1/blob/f7e01a9aed717856d75472614e3e53506a086466/cores/arduino/wiring_digital.c#L52), even if PWM was not used at all. Uncomment those 2 lines, and the size will go down.

One solution is not to call these 2 directly, but through a pointer that is set in pwm_start() / dac_write_value(). A little ugly.

rogerclarkmelbourne, 2016-11-16:
Looking at those calls, I thought perhaps the problem was that the GPIO_TypeDef * port is not constant, (it could be declared as const as its not modified, but I see now that the problem is that the g_analog_config array (of structs), is being updated inside that function, so the compiler has no choice but to put it in RAM

But looking at what pwm_start does. Its just using the timerHandle part of that stuct to pass into the HAL functions.

Unfortunately its also used in pwm_stop, hence why its saved for all pins.

However, I don't know if any of the HAL functions that pwm_stop call, e.g. HAL_TIMEx_PWMN_Stopor  HAL_TIM_PWM_Stop, and  HAL_TIM_PWM_DeInit really need to have valid parameters for the init part of TIM_HandleTypeDef as we are not init'ing the timer

We can still access the .instance by using 

g_analog_config[id].timInstance; 

because we have the id from 

int8_t id = get_analog_instance(port, pin);

I guess I'd need to try this and see if it works.

BTW. Do you know where the STM32 HAL API docs are ??

rogerclarkmelbourne, 2016-11-16:
I've had another look at this, and I think we may need to split these global structs into separate structs.

One which is constant and one which needs to be updated by the core.

danieleff, 2016-11-17:
> BTW. Do you know where the STM32 HAL API docs are ??

[Here](http://www.st.com/content/ccc/resource/technical/document/user_manual/72/52/cc/53/05/e3/4c/98/DM00154093.pdf/files/DM00154093.pdf/jcr:content/translations/en.DM00154093.pdf#545), But I just read the [source code](https://github.com/stm32duino/Arduino_Core_STM32F1/blob/master/system/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_tim.c#L1117).
I agree with separating const part. Although for example `timConfig` is always the same for every pin, it could be hardcoded into pwm_start?
The `g_analog_config` can be split in an other way too: 3 different structs for ADC/DAC/PWM. That cuts its size in less than half.

rogerclarkmelbourne, 2016-11-17:
Thanks

As far as I can see most of the data in the struct is static,

It looks like the data needed could be reduce to a few hundred bytes.

rogerclarkmelbourne, 2016-11-17:
Actually, this will be a serious problem on the V and Z series. They have masses of pins, so it will waste loads of RAM.
They do have more RAM but it if someone bought a Z series for the RAM and not the GPIO they would have problems with the current scenario

rogerclarkmelbourne, 2016-11-18:
Daniel

I may have a fix for this, but there is also another issue which I will explain later

See  https://github.com/stm32duino/Arduino_Core_STM32F1/commit/719b61b821672857da229a73e9e451f70036e7e8

I've declared `g_analog_config` as const and I've changed pwm_start and pwm_stop so that they don't need to update `g_analog_config` 

However, I can't get analog write to work at all, even before this change :-(

The Fading example seems to crash, and analogWrite just seems to be binary, on or off on the LED.

danieleff, 2016-11-18:
Which board/pin did you try to use?
Not all PWM pins are configured. Try one marked GPIO_PIN_PWM in variant.cpp (AND is also in g_analog_config). For example here is a [patch](https://github.com/stm32duino/Arduino_Core_STM32F1/files/599087/maplemini_PB1_PWM.patch.txt) to use Maple Mini LED. I do not advice to commit this, just to try it.

rogerclarkmelbourne, 2016-11-18:
I tried several pins, including the ones on the Nucleo which claimed to be PWM (pin 11 has PWM written next to it)

I wonder why not all pins are defined as being capable of PWM ?

I'll try your patch

PS. 

I'm not sure if this explains why running the "fading" example crashes the Blue Pill

rogerclarkmelbourne, 2016-11-18:
BTW. I raised a separate issue

rogerclarkmelbourne, 2016-11-18:
I've updated the WIP branch, it still does not seem to setup PWM correctly, but its no longer crashing ;-)

I'm not sure what's different in pwm_start between my code and the original code.

One problem I can see in the code is pwm_start return type is void, so it can silently fail.

danieleff, 2016-11-18:
When timHandle was global, it was initialized, but now it is not.
In `pwm_start()` you need `timHandle.State = HAL_TIM_STATE_RESET;`. (HAL_TIM_STATE_RESET = 0x00)
For reference, [here is the struct](https://github.com/stm32duino/Arduino_Core_STM32F1/blob/master/system/Drivers/STM32F1xx_HAL_Driver/Inc/stm32f1xx_hal_tim.h#L286). Channel, Lock, (hdma) may need to be initialized too.
Also, this is actually used to store state between HAL calls, so it might not be good to recreate itevery call... Handle it with care.

rogerclarkmelbourne, 2016-11-18:
Ok. You mean, use ..
  TIM_HandleTypeDef timHandle={};

That fixes it.

I did also comment out

``if(do_init == 1) 

So it always inits.

But I agree, its probably not a good idea, because I don't know whether the code in pwm_stop 
i.e 
HAL_TIMEx_PWMN_Stop  or 
HAL_TIM_PWM_Stop  

and

HAL_TIM_PWM_DeInit

Need anything that was stored in  timHandle

To be technically correct, we would need to store the whole timHandle struct.

So I've done an interim fix of having a separate array for just that struct, which saves just under 1k
See the latest WIP branch commit

To save more memory I can think of 2 options, but one of them is not technically correct.
1. Setup TimHandle each time pwm_start is called, and re-init the struct before passing to the HAL functions in pwm_stop.
I think this may work, and would save 1k of RAM, but we probably can't do this because its not correct use of the HAL.

2. dynamically allocate memory for TimHandle as required.

rogerclarkmelbourne, 2016-11-19:
In the WIP branch I changed it to move the timHandle to a separate array

I think this can be improved if I just have an array of pointers to the timHandle, then use malloc() in pwm_start of timHandle[x] = null

But do not use free() in pwm_stop() as that could potentially cause memory fragmentation if someone kept changing pinMode so that pwm_start(); then pwm_stop(); were called repeatedly

rogerclarkmelbourne, 2016-11-24:
I've removed the code that used malloc and have used the same method as used by mbed, except local instances of TIM_HandleTypeDef are used in pwm_start and pwm_stop

rogerclarkmelbourne, 2016-11-24:
Edit.

Although the code is working, I made a mistake and I don't know why its even compiling

g_analog_init_config.timInstance; 

does not exist.

I'll fix asap

rogerclarkmelbourne, 2016-11-24:
OK
I think its fixed now.

closed 2017-10-24 by fpistm