fpistm/Arduino_Core_STM32F1 #26: analogWrite does not work correctly

rogerclarkmelbourne, 2016-11-18:
Steps to reproduce

Install original version from boards manager package
Load example sketch Analog->Fading
change ledPin from 9 to 13
Upload

On Nucleo F103RB this seems to just flash the LED and not fade it, I checked with a logic analyser and it does not appear to deliver PWM on that pin

I tired on a different board on a different pin and got the same result

rogerclarkmelbourne, 2016-11-18:
Edit

Looks like perhaps this is "by design"

PWM other pins just gives high or low

However I don't know why PWM is available for most pins ?

danieleff, 2016-11-18:
I think that is because those pins support hardware PWM. The other pins would need to use software interrupt. How is analogwrite done in libmaple?

rogerclarkmelbourne, 2016-11-18:
I'm Not sure how libmaple does it.

rogerclarkmelbourne, 2016-11-18:
Fading on the Black Pill LED PB12 does not work, but I thought it worked on the Blue Pill Led on PC13, but my Blue Pill now had a broken USB connection on the PCB :-(

So I'll need to connect STLink to test

danieleff, 2016-11-18:
The following pins are currently set to (hardware) PWM: PA7, PB3, PB4, PB6, PB10, PC7
Here is a [list of possible PWM pins](https://gist.github.com/danieleff/fb7987d7e0dae6dbbbfb58beb87b1096), TIM (PWM) column.

danieleff, 2016-11-18:
There is one more thing that related to this: In `analogRead` and `analogWrite` there is a line: `uint32_t apin = ulPin&0x0000000F;`

It is only there because the size of g_anInputPinConfigured is only MAX_ANALOG_IOS. I guess it was copy-pasted accidentally to `analogWrite`.

Both of these lines should be removed, change size of g_anInputPinConfigured to MAX_DIGITAL_IOS, and take care of the increased ram size it in #27

fpistm, 2017-10-24:
Fixed in https://github.com/stm32duino/Arduino_Core_STM32

closed 2017-10-24 by fpistm