fpistm/Arduino_Core_STM32F1 #27: There is scope to reduce the RAM requirement by various changes

rogerclarkmelbourne, 2016-11-18:
I think RAM us can be reduced in a number of places - not just the g_analog_config 

''typedef struct _PinDescription
{
  uint32_t arduino_id;
  uint32_t ulPin ;
  GPIO_TypeDef  * ulPort;
  uint32_t mode;
  bool configured;
} PinDescription ;''

Seems wasteful as arduino_id would fit in uint8_t, pin would definitely fit in uint8_t 
mode would fit in uint16_t 
I don't know what size bool could convert to, but its going to take more than 1 bit, so the compiler may decide to make that a uint32_t (I'd need to check if this was worth changing to uint8_t to guarantee lowest memory usage)

In fact the pinDescription struct is used in both const and non const arrays.

Looking at 

```PinDescription g_intPinConfigured[MAX_DIGITAL_IOS];

The only part of this array that gets updated is the .configured

g_intPinConfigured[pin].ulPort
and 
g_intPinConfigured[pin].pin 

are read, but never written

So it would be more efficient to have another struct for the Interrupt in descriptors like

typedef struct _IntPinDescription
{
  PinDescription *pinData;
  bool configured;
} IntPinDescription ;

and instead of 
g_intPinConfigured[pin] = g_APinDescription[i]; 

use

g_intPinConfigured[pin].pinData = &g_APinDescription[i]; 

(I'm not sure if that would compile, you you get the idea ;-)

danieleff, 2016-11-18:
If #14 is fixed, where I mean PinDescription.arduino_id == index in the array, this becomes even more easier, because as you said only the bool configured is ever used/written to:

- `g_digPinConfigured`, `g_anInputPinConfigured`, `g_anOutputPinConfigured`,  `g_intPinConfigured` can be changed from PinDescription to simple bool array (or later bit array)
- PinDescription.arduino_id removed
- PinDescription.configured removed

rogerclarkmelbourne, 2016-11-18:
I think we may still need a pointer to pin in g_anInputPinConfigured because this array is smaller than MAX_PINS

The other option is to make all the arrays the same size i.e MAX_PINS, but this may take more RAM than having a pointer to pin struct  as well as the bool

(Code size is a bit bigger using the pointer to pin struct but RAM is very small and needs to be conserved, and there is not so much of a problem with flash )

stevstrong, 2016-11-20:
And what about changing the whole structure to be an array of pointers:
`PinDescription * g_intPinConfigured[MAX_DIGITAL_IOS];`
The array "g_APinDescription[]" should be then stored in FLASH (having removed the member "configured"), the others like "g_intPinConfigured[]" will be placed in RAM.
"#.arduino_id" is array index.
One could then make the member "#.configured" equivalent to whether the pointer stored in "g_intPinConfigured[]" is NULL or has been written with a non-zero value.

rogerclarkmelbourne, 2016-11-20:
@stevstrong 

Good idea. Saves 1 byte per array index, as in my solution we'd need to store the pointer to the pin description array item, so we may as well use null to indicate its not configured 

I've already  converted the array that stores the timer Handles into an array of pointers - but that was so I could use malloc - and I'm still not sure is people will accept that particular solution

rogerclarkmelbourne, 2016-11-20:
OK.
I just committed the change for g_intPinConfigured to the WIP branch

https://github.com/stm32duino/Arduino_Core_STM32F1/commit/6b552a91b7d9ae3b2ae9ffbb6ce7c1ae962030df

The configured property was was written but never read, so I commented out references to it, and no other code was needed to replace that functionality

stevstrong, 2016-11-20:
Don't know how much bytes we saved with this, but it certainly looks nice. :)

rogerclarkmelbourne, 2016-11-20:
I tried checking out the master branch and then checking out the WIP branch on my machine to see if I could compare, but somehow it didn't make any difference.

I'll install the JSON package on a different machine and compare the Nucleo with the original version

rogerclarkmelbourne, 2016-11-20:
At the moment, I'm still not sure about my solution for the timerHandle, as using malloc could cause the sketch to run out of RAM even though the IDE reports they have enough RAM.

fpistm, 2017-10-24:
Several RAM size consumption done in https://github.com/stm32duino/Arduino_Core_STM32

closed 2017-10-24 by fpistm