fpistm/Arduino_Core_STM32F1 #28: analogRead(pin) overrides the pin it is given

danieleff, 2016-11-18:
In original Arduino analogRead treats pin A0 (= 14) as 0.
The current code does that in wiring_analog.c pinConvert(), that converts pins A0 (= 16) to 0 etc...

This conversion is only needed in Nucleo variant to emulate the arduino board, and will cause confusion in other boards. In bluepill calling analogRead(PA0) will try to read pin PA0 | A0 => PB0

One way is to make ARDUINO_PIN_A0 = 0 in other variants, which is confusing.
Other is to pinConvert to variant.cpp, and just return ulPin in non nucleo boards.

rogerclarkmelbourne, 2016-11-18:
Can we just change g_APinDescription for the Nucleo so that index 0 = pin 0 on the Nucleo 

Thats what libmaple does e.g.

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/variants/nucleo_f103rb/board.cpp

with 

```
enum {
    PA3, PA2, PA10, PB3, PB5, PB4, PB10, PA8,
    PA9, PC7, PB6, PA7, PA6, PA5, PB9,PB8,
    PA0, PA1, PA4, PB0, PC1, PC0,
    PC10,PC12,PB7,PC13,PC14,PC15,PC2,PC3,PC11,PD2,PC9,PC8,PC6,PC5,PA12,PA11,PB12,PB11,
    PB2,PB1,PB15,PB14,PB13
};
```

vs

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/variants/generic_gd32f103c/board.cpp

As most boards do not have the strange mapping, and we can do a 1 to 1 lookup just using the array index

I think its better to map it in the same way as Libmaple

danieleff, 2016-11-20:
This is not _that_ important right now, just don't test if pins are correct using analogRead()

Problem is in [pinConvert](https://github.com/stm32duino/Arduino_Core_STM32F1/blob/master/cores/arduino/wiring_analog.c#L90), and cannot be fixed with pin numbers.

On Nucleo board the following do exactly the same, and it is correct (compatible with arduino): `analogRead(0)` == `analogRead(A0)` == `analogRead(16)`, they will all read the PA0 pin (untested a I don't actually have that board).

However on BluePill:
```
void setup() {
  Serial.begin(9600);
}
void loop() {
  Serial.print(analogRead(PA0));
  Serial.print(" ");
  Serial.println(analogRead(PB0));
  delay(100);
}
```
this will both read from pin PB0, which is wrong.

rogerclarkmelbourne, 2016-11-20:
Did this ever work?

rogerclarkmelbourne, 2016-11-20:
One solution is to move pinConvert into variant.cpp

Perhaps call it analogPinConvert as pinConvert seems to only be used in wiring_analog.c and only used once. (so in its current form it doesn't even need to be a function)

We could remove the enums etc for the real port / pin values, so that they can only access analogRead() on a nucleo using either a number or the ARDUINO_PIN_A0 type names

I'm not sure who came up with names like ARDUINO_PIN_A0  or whether they were intended for internal use in the core or for users.

It doesn't look like mbed or Arduino.cc use them, so I think perhaps they are internal names only.

So the user would be expected to use analogRead(0); 

In which case if we have move pinConvert (and give it a new name) in variant.cpp, that should resolve the problem (assuming we can get rid of PA0 etc, which I suppose could be confusing on the Nucleo)

fpistm, 2017-10-24:
Fixed in https://github.com/stm32duino/Arduino_Core_STM32

closed 2017-10-24 by fpistm