fpistm/Arduino_Core_STM32F1 #35: I2C pins are hard coded in twi.c so only work with Nucleo hardware

rogerclarkmelbourne, 2016-12-09:
Pin usage for I2C is hard coded in twi.c

https://github.com/stm32duino/Arduino_Core_STM32F1/blob/master/cores/arduino/libstm32f1/twi.c#L148-L176

As the Nucleo uses remapped pins for I2C the initialisation for this should be in variant.cpp so that it can be configure on a per board basis

danieleff, 2017-01-10:
I created an example here: https://github.com/danieleff/Arduino_Core_STM32F1/commits/I2C_variant
with the most important commit this one: https://github.com/danieleff/Arduino_Core_STM32F1/commit/52b193bec8738775dfe9b2c574f89481810df901

1. This is based on #41
2. I only tested on Maple Mini I2C1, default pins
3. I was not able to put the `i2c_instance_e` enum into the variant.h, it just gave compile errors. I had to put it into a new `variant_peripheral.h`. Need help with this one!
4. (__HAL_RCC_AFIO_CLK_ENABLE() can be removed at later point)
5. (`GPIOB, GPIO_PIN_7` could be simplified to `PB7`, but its good enough)
6. (There could be a i2c_config_pins(...), so the user can change to alternate pins. But I bet it would be really really rarely used)

rogerclarkmelbourne, 2017-01-11:
OK.

I presume you just meant

```
#define I2C_SDA_PORT GPIOB
#define I2C_SDA_PIN GPIO_PIN_9
```

etc

danieleff, 2017-01-11:
Sorry I forgot to say that it was the variation where the whole structure (reduced in size) is moved into the variants. I did not write an example for the #define variation, as that's literally just 5 #define in `variant.h`.

@fprwi6labs you wrote you are looking into this too, do you have a suggestion?

bperrybap, 2017-01-12:
Ok, so this directly relates to an issue I was about to open but I think it may actually be better to put it here.
My issue is I have a diagnostic sketch in my hd44780 that uses the symbols SDA and SCL which come from the variant file and it won't compile because the stm variants don't define these pin symbols.
EVERY single pins_arduino.h file and I really do mean EVERY one, on every processor and every core on every IDE version I've used to to date (and I have used many IDEs on many different processors and cores over the past 8 years), defines these two symbols in the pins_arduino.h file.
They represent the digital pin numbers for the SDA and SCL pins that a sketch can use with the digital API functions pinMode(), digitalRead(), digitalWrite().

So my recommendation would be create the SDA and SCL symbols in each for variants and use them in the twi module. OR.... define the i2c pins in a hal file and then pull in those definitions to define SDA and SCL in the pins_arduino.h file so that SDA and SCL exist for the sketches.

It looks like  each pins_arduino.h file seems to include variant.h which doesn't seem to have any C++ code so even though the twi module is being compiled as C it should be able to include the pins_arduino.h header file.

If this twi code has to build in non arduino environments as well or it is preferred to define the pins in a hal type file then it might be better to put the i2c pins in hal header file that gets included by twi.c and have twi.c get the pin number from the hal header rather than hard code it.

However it is handled, all the variants should create the symbols SDA and SCL so that sketches can have access to the default digital pin numbers for the i2c pins.

rogerclarkmelbourne, 2017-01-12:
@bperrybap 

That's an interesting point.

We can define a pin reference (number) for SCL and SDA, but I'm not sure how whether the compiler will be able to do a lookup in the g_APinDescription array to convert those pin numbers into the Port / Pin data that is needed for the I2C init struct

If that is possible, we'd also need to add the pin "mode" as used here (and on line 169)
https://github.com/stm32duino/Arduino_Core_STM32F1/blob/master/cores/arduino/libstm32f1/twi.c#L164

Because the Nucleo board doesn't use the default / normal pins for I2C it uses the AF (alternate function) on the pins in question 

So we need to 
(a) Add defines for SCL and SDA
(b) try changing g_i2c_init_info struct to use data from g_APinDescription and see if that actually complies and works
(c) possibly add a new element to the g_APinDescription to handle the AF 
Note. I see that g_APinDescription already has a property/element named "mode" but I'm not sure what its already used for, and whether it can be used for this purpose.

danieleff, 2017-01-12:
(a) Agree
(b) This should be easy
(c) sda_mode/scl_mode is always GPIO_MODE_AF_OD when using I2C, which is why I removed it in PR #41. The alternate function selection is done in .i2c_alternate function pointer. This is the only complication.

rogerclarkmelbourne, 2017-01-12:
@danieleff 

We'll need to double check (b) is OK. I think it should be, but we had issues with this sort of thing in the LibMaple core if the source struct got put into RAM by the compiler .
Hopefully in this case g_APinDescription is in Flash as its const, so it should not be an issue.

Re: (c)
OK, I confused the I2C function setup of the pin (it not being GPIO) with the alternative I2C mapping, which like you say is done by this function

https://github.com/stm32duino/Arduino_Core_STM32F1/blob/master/cores/arduino/libstm32f1/twi.c#L145

And I can't think of a good / clean way to handle this.  e.g. we could use something like 

> #define I2C_AFIO_REMAP

Then use an ifdef in that function, but that is a messy solution :-( , really large parts of this code need to be moved to the variant folder, but that would be a big change

bperrybap, 2017-01-12:
large parts of what code?

rogerclarkmelbourne, 2017-01-12:
@bperrybap 

I tried moving the whole static i2c_init_info_t g_i2c_init_info[NB_I2C_INSTANCES] struct to the variant, but it was a complete can of worms, as that struct references the setup functions in twi.c and also there were a load of issues with includes

So it was a waste of time trying to do it that way.
I think @danieleff  looked at doing something similar and came to the same conclusion.

Ideally the config for the I2C should be simple to setup / define for each variant, but the way the code has currently been written, there doesnt seem to be a simple and clean way to do that

bperrybap, 2017-01-12:
@rogerclarkmelbourne 
I don't follow what you are doing.
So I don't see the value nor the need to move the actual struct table from where it is.

You don't have to move the table into the variant file to get a value/values defined in the variant file or a valued defined off in some other hall file, into that table.

The big question is whether that i2c code needs to be buildable outside of the arduino environment.
If so, then it still seems possible but it will involve doing things a bit differently than what could be done in an arduino environment.

Also if there are needs to have that table using constants because there are some games being played by having access to certain values at compile time, there are ways to deal with that as well. But the table must be declared not only as static but const as well.

I'm still not seeing the problem.
I guess I need to actually install the core to see what is going on and how it is used to see what needs to be done to fix it.
(I just don't have any h/w right now to test it)

bperrybap, 2017-01-12:
In my mind, worst case, there is a conditional that extracts values from the variant file or uses defaults (the values currently being used) depending on if the code is being built using an Arduino environment.
I'm off to do some poking on the code.

fprwi6labs, 2017-01-16:
@danieleff 
Sorry for the late answer but we was really busy last week. 

We have a dilemma between decrease the ram usage and the configuration flexibility. Currently we don't have better solution than you. We just try to hard code the init structure inside a file which depends on the variant to give more flexibility without increase the ram usage.
We pay attention to your suggestions.

fpistm, 2017-10-24:
Done in the https://github.com/stm32duino/Arduino_Core_STM32

closed 2017-10-24 by fpistm