fpistm/Arduino_Core_STM32F1 #43: Tools (win) folder possibly wrong in platform.txt

palmerr23, 2017-04-05:
Generally works fine on my Maple Mini. Thanks ;-)
Nice to see the Cube files in action and tiny variant folders!

I needed to change
{runtime.tools.STM32Tools.path}
to
{runtime.hardware.path}
in 
platform.txt 
to get the DFU bootloader to work (java can't find file error on attempted upload).

runtime.tools.STM32Tools.path doesn't appear to be defined. (Maybe my config issue, but others might have similar setups.)

Arduino 1.6.12

New Cube STM32F1 files placed in C:\Program Files (x86)\Arduino\hardware\STM32\ alongside all my working "Maple style" STM32duino  files with a common \tools and \drivers folders.

fpistm, 2017-04-05:
Hi,
have you installed it thanks the boards manager? or do you use the repo?
If you use the repo, the error is normal as the STM32Tools is not know. It is installed as a dependency thanks the board manager.
That's why I've created a stubpackage:
https://github.com/fpistm/boardManagerStub

danieleff, 2017-04-05:
Why not just rewrite in the packager script from {runtime.hardware.path} to {runtime.tools.STM32Tools.path} when creating the json?
That way developers can use the repo, and users the boards manager.
The stub package is not needed then.

fpistm, 2017-04-06:
good idea @danieleff . It should be fine like that.

fpistm, 2017-10-24:
Done in https://github.com/stm32duino/Arduino_Core_STM32

closed 2017-10-24 by fpistm