fpistm/Arduino_Core_STM32F1 #44: Serial3 not found ?

hazmi-e205, 2017-05-23:
I have a BluePhill board STM32F103C8t6. 
I can't use UART3 when i tried to use this Core.
[image]

Actually, on STM32CubeMX i found the USART3. Check this..
[image]

fprwi6labs, 2017-05-23:
@hazmi-e205 could you try the patch #45 . Let me know if there is any problem.

hazmi-e205, 2017-05-23:
Sorry @fprwi6labs , I'm newbie...
On my Arduino Board Manager not available update file.

but I tried copy manually in directory: /home/hazmiarifin/.arduino15/packages/STM32/hardware/stm32f1/2017.1.20

And Still....
[image]

fprwi6labs, 2017-05-23:
Sorry  @hazmi-e205 I didn't generate new package for the Board Manager yet.

Just to be sure, did you replace uart.c & uart.h in cores/arduino/libstm32f1 and variant.cpp & variant.h in variants/STM32F103C_BluePill with the patched files?

Please checks that you find the Serial3 definition in .arduino15/packages/STM32/hardware/stm32f1/2017.1.20/variants/STM32F103C_BluePill/variant.cpp & variant.h.

hazmi-e205, 2017-05-23:
Oke, it's compiled well. 
[image]

I hope work fine too on my board.
I will try tomorrow.

Thank you for your help @fprwi6labs

fprwi6labs, 2017-05-23:
Fine! 
Please let me know if Serial3 is completely functional.
Enjoy it!

hazmi-e205, 2017-05-30:
On my board, I used multiserial example for checking received  data. But the result is...
[image]

if i used "Generic STM32F103C" from @rogerclarkmelbourne repository. Work fine...
[image]

fprwi6labs, 2017-05-30:
@hazmi-e205 
It seems to be an mounting problem with the serial port.
Could you give us more information (flashing method, OS, ...)?

hazmi-e205, 2017-05-30:
@fprwi6labs 
I'm using STLink for flashing my board.
my OS is deepin 15.4 (Debian Based)

I think you right, but I tried to check "lsusb" my board not listed
[image]

If the problem is mounting the serial port on my PC,  i tried simple program serial usb that work fine
[image]

fpistm, 2017-10-24:
No update. 
Patch from @fprwi6labs  merged.
Anyway, this core is deprecated instead use https://github.com/stm32duino/Arduino_Core_STM32

closed 2017-10-24 by fpistm