fpistm/Arduino_Core_STM32F1 #49: Board reboot without pressing a reset button.

Adminius, 2017-06-09:
Hi,

i founded this solution:
http://www.stm32duino.com/viewtopic.php?t=678
`nvic_sys_reset();`
but this function is not included. 
Do i have another possibility to reset my board within software?

rogerclarkmelbourne, 2017-06-09:
I'm not sure what the HAL way to do this is

That function is actually this code, which uses direct access to registers

```
void nvic_sys_reset() {
    uint32 prigroup = SCB_BASE->AIRCR & SCB_AIRCR_PRIGROUP;
    SCB_BASE->AIRCR = SCB_AIRCR_VECTKEY | SCB_AIRCR_SYSRESETREQ | prigroup;
    asm volatile("dsb");
    while (1)
        ;
}
```

fpistm, 2017-06-09:
Hi @Adminius,
Just call:
NVIC_SystemReset();
it is not link to HAL. It is a CMSIS interface. It is available in core_cm3.h.

bilbolodz, 2017-06-09:
Maybe you should use watchdog functionality?

Adminius, 2017-06-09:
Maybe... :)
But how? ;)

bilbolodz, 2017-06-09:
Not checked just google it: http://www.stm32duino.com/viewtopic.php?t=140

rogerclarkmelbourne, 2017-06-09:
@bilbolodz 

Most of the stuff on the forum relates to the LibMaple core, not this new HAL based core.

Re: Using the WDT to reset..

Well I suppose you could do it that way, but if NVIC_SystemReset() exists in the CMSIS, I don't see any reason not to use it

Adminius, 2017-06-29:
tested with L476RG. it works. 
still waiting for BluePill :(
solved for me.

closed 2017-06-29 by Adminius