fpistm/Arduino_Core_STM32F1 #51: PWM resolution and other pin related settings

Adminius, 2017-06-19:
Hi,
i'm confused about 8bit PWM resolution...
https://github.com/stm32duino/Arduino_Core_STM32F1/blob/master/variants/STM32F103C_BluePill/variant.h#L143
Is there a reason for it? or can i make a PR to increase it to 16bit?
Why is PWM supported only on few pins? (PA7,PB3,PB4,PB6,PB10)?
PWM can be used on at least 12 pins...

https://github.com/stm32duino/Arduino_Core_STM32F1/blob/master/variants/STM32F103C_BluePill/variant.cpp#L39
There are only A0-A3 available. But PA2,PA3,PA5,PA6,PA7,PB1 can also be used for ADC.

[image]

fprwi6labs, 2017-06-20:
Hi @Adminius 

Yes you can increase the PR to 16bits but I not sure about the final result. It depends on the final timer resolution.

The STM32F1 repository will be merged soon inside https://github.com/stm32duino/Arduino_Core_STM32. It will be easier to configure all pins of a board.

I invite you to start to look at the generic repository to see the new pin management.

Adminius, 2017-06-20:
Timer 1-4 are 16bit timer. With 72Mhz CPU Clock and 16bit resolution it's abour 1.1khz PWM frequency.
i've made similar PR for SAMD21 core:
https://github.com/arduino/ArduinoCore-samd/pull/226/commits/a5e0003a7b85a7cf0dd614f55c7b3a8c1bbcec6d#diff-63b7293bbdf8f8f0f8b98336a5e85d83L215

the same way i would made it for STM32:

- Standard resolution 16bit

- All other resolution will be mapped to 16bit with mapResolution.

But I can wait until both cores will be merged...

fprwi6labs, 2017-06-21:
There is a limitation for the moment. The resolution is limited to 8bits.
We will fix this problem as quick as possible.

Adminius, 2017-06-22:
I think, it should be enouth to change this lines 
https://github.com/stm32duino/Arduino_Core_STM32F1/blob/master/variants/STM32F103C_BluePill/variant.h#L143
to:
#define PWM_RESOLUTION                16

#define PWM_FREQUENCY                 1000

#define PWM_MAX_DUTY_CYCLE            65535

fprwi6labs, 2017-06-23:
@Adminius 

Did you test with this modification? Did you obtain the result you wanted?

Just for information: the STM32F1 merged inside the generic repository should be available next week.

Adminius, 2017-06-23:
not tested yet. still waiting for my test board :(

next week sounds good 👍

Adminius, 2017-06-29:
just tested this:
#define PWM_RESOLUTION 16
#define PWM_FREQUENCY 1000
#define PWM_MAX_DUTY_CYCLE 65535

on L476RG with 80Mhz
PWM frequency is 1.2 kHz (with 72MHz it will be 1.1 kHz)

user can use analogWriteResolution(res) to change the resolution.

fprwi6labs, 2017-06-30:
Thank you @Adminius 
I forgot to use the function analogWriteResolution while I was testing.
I will try again.

fprwi6labs, 2017-07-05:
It would be better to replace PWM_RESOLUTION and PWM_MAX_DUTY_CYCLE with variables like _writeResolution to set them directly inside analogWriteResolution().
I will add the modification when the F1 will be merged inside the generic repository.

Adminius, 2017-07-13:
This ticket will be reopened in general repository .

closed 2017-07-13 by Adminius